Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #71: Tuner cannot match any antenna and Effiency stays at 33 percent

sridharkr, 2021-09-14:
I have the ATU-100 and after connecting an antenna and working on the tuner, the reset works and post which the antenna is never able to tune perfectly. It is always at 33 percent efficiency and shows SWR 9.99 , the C values and L values keep changing often but still never is able to tune a perfect match.

Can you give some guidance as to why this would be the case and what can I do to check this behaviour against?

Thanks

[image]

Appreciate your response

Dfinitski, 2021-09-14:
Hi
Check if the power meassured correctly, if not try to change the tandem
match coil phase.

killerekpl, 2023-05-31:
@sridharkr Did you solved the problem? I have the same problem