Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #73: Unexplanaible behavior of device going to bypass mode by itself

mautas, 2021-09-30:
Hi! I am experiencing a weird problem with 7x7 device. After some weeks of just storing the device (it was working ok) I took it out. to use it.. When I turn it on it switches to bypass mode by itself. All buttons are unresponsive after startup. There are 100 nF capacitors in secondary buttons. Device can enter both test modes, and Feeder Line loss setup mode, but after that buttons are unresponsive.

Checked out for shorted capacitors, found none.

Dfinitski, 2021-09-30:
Hi. Check the voltage on the shorting capacitors when you press the
buttons. It should be 5V when unpressed and 0V if the button is on.

mautas, 2021-09-30:
Reading almost 0V (0.04V) at RB1, the cable from the RB1 point at the board to the button looks ok but I'll check it again..

mautas, 2021-09-30:
> Reading almost 0V (0.04V) at RB2 the cable from the RB2 point at the board to the button looks ok but I'll check it again..

Cable was OK, I even unsoldered it... RB2 at low value... Faulty PIC?

Dfinitski, 2021-09-30:
Should be 5V and 0V when button is pressed.
Your 0 V in what position ?

mautas, 2021-10-01:
The tuning pad (B2) is fixed fixed at 0.04V... seems like something went wrong there blowing up that pin, I wonder if is some RBx pin unused so I could remap RB2 to that pin (of course I'd have to wire it) in order to save the PIC. I'd also have to compile the project to honor the new mapping.

mautas, 2021-10-06:
Well I could save the PIC... it seem like the pullup circuitry for B2 blowed up or something. I tested feeding 5 volts to B2 and the signal was ackowledged by the PIC, so I finally connected the button to 5 volts through a 330 ohm resistor after trying out different resistor values from 10K to 1K. I think it was not a good idea to rely on internal pullup circuitry of the PIC...