Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #78: I/O short-circuit

OK1TK, 2021-11-07:
Hello, I just finished the board. I don’t want to use the SMA connectors and I soldered a few  cm of RG58 to the Input and Output. When I checked the cables with multimeter I found short-circuit between the signal and ground. I thought I did something wrong with the cables but I have the same situation when I removed the cables. I can’t find any mistake I was checking all the SMD components for short-circuit when each of them was soldered.
The power supply was not connected yet and any button nor the display are not attached yet.

Please, do you have any recommendation what to check what are the common mistakes ?
Thank you in advance.

Dfinitski, 2021-11-07:
Hi.
See the schematic. RF input is shorted to ground with the tandem-match.

OK1TK, 2021-11-07:
Hi, I  was looking to the schematic and just found it 2 minutes before your
e-mail arrived. Thank you very much for your fast reply.

May I have another question ? Pls, do you think I can connect the display
and buttons via a few mezery of UTP cable for “Remote” control ?

Have a Nice day.
73!
Tomáš