Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #81: What is power losses in bypass mode?

qrp73, 2021-12-23:
Hi, what is typical power losses (S21) in bypass mode for extended 7x7 version?

I'm using tandem match 1:10 and as I found there are two bypass modes:

1) device is powered off (all relays are OFF)
- 1 MHz S21 = -0.12 dB
- 7 MHz S21 = -0.17 dB
- 14 MHz S21 = -0.25 dB
- 30 MHz S21 = -0.48 dB

2) device is powered on and bypass mode enabled (S8 relay is ON and the rest relays are OFF)
- 1 MHz S21 = -0.14 dB
- 7 MHz S21 = -0.19 dB
- 14 MHz S21 = -0.27 dB
- 30 MHz S21 = -0.44 dB

So, there is 10% of power loss at 30 MHz. For 100W TX at least 10W will be dissipated on ATU100 in bypass mode. This looks pretty high losses, so I wonder if there is something wrong with my device or that results are normal?

Dfinitski, 2021-12-23:
Hi. What about SWR on 50 Ohm dummy load ?

qrp73, 2021-12-23:
When output is loaded with dummy load ATU100 shows SWR about 1.00 .. 1.02.  

Second power meter connected on ATU100 output shows at 7 MHz about 5% less power (-0.2 dB loss) than connected directly to the transmitter.

VNA shows S11 on ATU100 input the following:

1. device is powered off (all relays are OFF)
- 1 MHz S11 = -26.09 dB / SWR = 1.10
- 7 MHz S11 = -28.88 dB / SWR = 1.08
- 14 MHz S11 = -24.69 dB / SWR = 1.12
- 30 MHz S11 = -17.87 dB / SWR = 1.29

2. device is powered on and bypass mode enabled (S8 relay is ON and the rest relays are OFF)
- 1 MHz S11 = -26.04 dB / SWR = 1.11
- 7 MHz S11 = -28.93 dB / SWR = 1.07
- 14 MHz S11 = -25.82 dB / SWR = 1.11
- 30 MHz S11 = -23.89 dB / SWR = 1.14

Dfinitski, 2021-12-23:
Looks good.
Connect a real 100W transmitter to some gauge with and without the tuner. Do you see 10% loss ?

qrp73, 2021-12-25:
Yes, I see 12% power loss when compare direct dummy load and through tuner.

Dfinitski, 2021-12-25:
Excellent. Now you can find what is become warm. I think the tandem match core

qrp73, 2021-12-25:
yes, transfluctor heats up during TX. I'm using BN-43-202, got it from Aliexpress with ATU-100 DIY kit. 

Is it normal? or I got bad transfluctor?

Dfinitski, 2021-12-25:
I think that is an approximate analogue, not exact bn43-202. High quality and original parts you can find on kitsandparts .com

qrp73, 2021-12-25:
what is expected power loss on 30 MHz with original good quality BN43-202?

Dfinitski, 2021-12-25:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/81/1001020437@
> github.com>

I don't know. That is good question for our official conversation group on
groups.io