Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #82: L4 Core getting very hot

w0ati, 2021-12-23:
I had noticed after using the ATU, the case was a little warm. Its sitting on top of the radio so did not give it much thought.
Today after using it I opened it for other reasons. I got the "Allan Bradley" smell right off. So looked further. 
I found the core of L4 was very hot and the paint has burned off (hence the smell). Most of the tuning uses that coil and 
combinations of the air wound coils.
I am using 100W or less.
Is this expected??

Dfinitski, 2021-12-24:
No. Seems like you have bad cores with large losses.nothing should be burned or has some smell.

w0ati, 2021-12-24:
I swaped out the core and made a contact with a configuration that used L4 (the new core). It did not appear to get hot. I will monitor it for a while. I am now concerned about all the other cores. The tuner rarely uses any inductor higher than L4. So the other cores in this kit could be bad and I would not know it. I am wondering if I should buy all new cores from a known/dependable supplier.

Dfinitski, 2021-12-24:
Kitsandparts.com

nv1o, 2022-02-20:
Isn't it important to wind the cores in the same direction?  I didn't see it mentioned in the instructions, but would think it might be desirable.

Dfinitski, 2022-02-20:
There is no any assembling instructions. I have added couple photos on the schematic to clear questions like this .

nv1o, 2022-02-20:
Yes, those were the instructions I was referring to, coil size, number of turns, etc., and yet my question remains unanswered.

Dfinitski, 2022-02-20:
Ok, there is a photo of tandem match too.

soosp, 2022-04-19:
Same issue here:
[image]
[image]

w0ati, 2022-04-19:
Fascinating, at some level it's nice to know I am not along (crazy). Seems like the big disadvantage with
the implementation shown in the link is coupling between inductors. Note L2 and L3 are at right angles
to each other and L1 is place off somewhere else. This would be to lower coupling between them. The number of windings and loss (series R) would come in to play at some point.
To be honest, I have stopped relying so much on the ATU and just tune the antenna lengths.

Dfinitski, 2022-04-19:
That is famous problem connected with bad Chinese "amidon" cores with high losses on RF.
Change them https://kitsandparts.com/store2.php

Hobby4life, 2022-04-19:
I have measured these cores with the advised amount of windings, but none came nowhere at their given inductance. I have replaced these cores with original Amidon T68-2 cores.. and this looks more promising.