Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #83: ATU-1000

helikeith, 2021-12-24:
Are there any instructions on the build of the ATU-1000. I have purchased the kit. but no information came with it.

Dfinitski, 2021-12-24:
Hi. Ask the seller about that.

Hobby4life, 2022-06-24:
What do you want to know?

c0nsul, 2023-01-19:
> What do you want to know?

how to set up the measurement of SWR? there are two blue potentiometers, "forward" and "reflected". What is the setting mechanism?

PD5DJ, 2023-01-19:
> > What do you want to know?
> 
> how to set up the measurement of SWR? there are two blue potentiometers, "forward" and "reflected". What is the setting mechanism?

What I did is the following:

1. Connect TRX to INPUT of tuner and Connect Dummyload to OUTPUT of tuner.
2. Now transmit for example CW 20 Watt.
3. Adjust the FWD trimmer, so display shows 20Watt
4. Also, measure the voltage accros GND and FWD pin of the 3 pin connector, and write this down.. for example 1.30Volt.
5. Now connect TRX to OUTPUT of tuner, and connect Dummyload to INPUT of tuner.
6. Now transmit again with CW 20Watt.
7. Measure the voltage accros GND and REFL pin of the 3 pin connector and adjust it to 1.30Volts.

Thats it.

Now the REFL input knows the same Power as the FWD input for proper SWR calculation.

You can only do this with a 50ohm load.
Do not use a antenna for this.

c0nsul, 2023-01-20:
Ohh... God bless you dear Bjorn! I was able to experimentally adjust the power indicator, but adjusting the SWR and the entire system as a whole baffled me. Now, I think I have more understanding. Thank you very much for your advice!

c0nsul, 2023-01-21:
> You can only do this with a 50ohm load. Do not use a antenna for this.

I setup the SWR and power according to the technology described above at a power of 10 watts. Now I have the following situation.

ATU always shows good weather and does not work correctly.

Real SWR 1 - ATU 1, real 1.5 - ATU 1.1. Real 2 - ATU 1.5
Power as a whole shows correctly, but with such measurements it makes no sense. If the SWR is 2.2 it starts tuning...

The result of the ATU tuning is often incorrect, it gets a real SWR of 1.8, and at the same time it writes that it is 1.1 or 1.2, but the transceiver shows 1.8.

I met comments on the forums that the problem may be in the Chinese diodes D3 and D4 (1N5711), that it is better to replace them with other similar diodes with a smaller error and better quality...

Maybe I should have adjusted the SWR not to 10W, but to 20W or 30W? The voltage measured between the GND and REFL/FWD was only 0.2 V.

Does it matter in which tuner mode to take measurements? Bypass or AUTO? SWR is everywhere but maybe it's important?

Dfinitski, 2023-01-21:
Hi, Dmytry.
Do a calibration with a power 500 W or above.
Do a tuning with a power around 100 W

c0nsul, 2023-01-22:
> Hi, Dmytry. Do a calibration with a power 500 W or above. Do a tuning with a power around 100 W

Thank you very much for your advice, will try to repeat calibration again.

c0nsul, 2023-01-22:
Re-calibrated with PA on 200W, but still issue with SWR. Next my idea - incorrect rings in tandem match. I measured the inductance on them and the difference in readings was 3 times. I think that my problem is that the Chinese sent me rings with different permeability. I'll try to buy elsewhere and rework this part of the tuner.