Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #84: Usdr all mode 8 band hf

TA3GT, 2022-01-08:
Hello: How should the device be connected to the ft8 interface cable? need interface diagram

Dfinitski, 2022-01-08:
HI. RF cabel in, out to an antenna. That's it.

TA3GT, 2022-01-08:
Beni yanlış anladın,ft8 için PC bağlantısı arayüz sürekli tx yapıyor

TA3GT, 2022-01-08:
You misunderstood me, PC connection for ft8 interface is always tx, no vox

TA3GT, 2022-01-08:
Is there a circuit between the computer and the cable?

Dfinitski, 2022-01-08:
Sorry, I don't understand you, this device does not have a TX/RX commutation. Are you about ATU-100 antenna tuner ?

TA3GT, 2022-01-08:
usdx all mod 8 band hf radio ft8 interface i wanted computer connection sorry about that

Dfinitski, 2022-01-08:
You have to use USB to COM converter cabel for CAT control and 2 audio cables to connect audio and mic connector to appropriated connectors your sound card on PC.