Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #85: FW ver 3.2

phillx19090, 2022-01-28:
Apologies if this is incorrectly posted or already covered. Thanks for a great project. I updated today to firmware 3.2. All OK but I wondered if there was a significance with the dots and dashes on the bottom row of the 0.95" OLED? I have these on the display and noticed they appear on the images on a closed issue last year.
Kind regards
Phill EA5JHA

Dfinitski, 2022-01-29:
Hi. Dot - auto mode is on.
Dash - Bypass is enabled now. See the additional 2 buttons description.

phillx19090, 2022-01-29:
Hi David, perhaps I didn't make it clear. The dots and dashes are at the bottom of the display. On boot up they appear to "underline" the firmware version. Then under the C value. It can be seen on the image on closed issue 69 by dl1hbt.

Dfinitski, 2022-01-29:
Oh, now I got you. You have to shift the image up or you don't have to shift it. See the cell description for 1,2,3 cells .

phillx19090, 2022-01-29:
Hi David, thanks for not making me feel too stupid. Read the manual quite a few times and somehow missed cell 7. Dropped it from 01 to 00 and all good. Just a note, my aliexpress kit has an OLED where the address is either 7B or 7A. Zero ohms select soldered on 7B - Cell 00 works with 78 set but Cell 01 needs 05 to get the display to work. Please close the issue.

closed 2022-01-29 by Dfinitski