Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #86: Tuning not activated

bitslave, 2022-02-05:
Hi David,

I'm wondering whether you could point me in the right direction regarding issues with my unit purchased fully assembled. It will not activate tuning with 10W fed to the tuner from the transceiver and is showing very low power readings and efficiency in all modes - bypass/auto tune/manual tune.

Transceiver is showing SWR of 2.1:1 during transmit with the below readings on the tuner.

[image]

Any help/suggestions greatly appreciated.
 
Kind regards.

Dfinitski, 2022-02-05:
Hi.
Did you connect a 50 Ohm dummy load to the antenna connector on the tuner ?

bitslave, 2022-02-05:
Hi David,

Thanks for the reply. These readings are with EFHW 80-10 antenna connected to the tuner. I don't have a dummy load (still on back order).

Dfinitski, 2022-02-05:
If there is no a seal, open the box and check the wires , connectors. I would check also coaxial cables you use for shorting, as a first step.

bitslave, 2022-02-05:
Ok, will do... Thanks.

bitslave, 2022-02-05:
Both UHF chassis mount connectors have continuity between center pin and body (ground). Can't see soldering as they are covered by heatshrink. Tandem match pic also attached.

[image]
.

Dfinitski, 2022-02-05:
Right side on the foto has 11 turns, left side has 10 ???? Am I right ?
Anyway you have to have a dummy load to continue the troublecheck.

bitslave, 2022-02-05:
Looking from the other side, it seems 11 turns for both T1 and T2. Will organise a dummy load for further testing. Thanks again.

Dfinitski, 2022-02-05:
Should be 10 turns. I think this device was not even check out after assemly. 
I would check all the soldering on relays, bottom side. 
Most likely some nodes are missed.

bitslave, 2022-02-05:
Agreed. Both sides of the tandem-match were 11 turns, I have corrected them to 10 turns. Still the same issue. Will now investigate the soldering of the relays. Thanks again for your assistance. I have also discovered the ATU100@groups.io boards. Will do some research there. Cheers.