Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #87: Question about van 3.2

davidhay45, 2022-02-06:
“Added new feature to disable all relays along with the timer display. New cell 35(33 for mini board) - 01 to activate the relay off function. Display Off Timer setup - cell 32 (30 for mini board). You can use this feature in your QRP setup where power consumption is critical.”

what is this to be used for?

Dfinitski, 2022-02-06:
Hi. Bypass mode will be turned on when the display will be turned off.

davidhay45, 2022-02-07:
So in other words when this is function activated by setting the EEPROM cell then putting the tuner into bypass mode the relays will be deactivated and the display will be turned off.

Dfinitski, 2022-02-08:
No.
Display has a timer for turning it off after some period, an example 15 minutes.
Is this clear ?

davidhay45, 2022-02-08:
OK, thnaks

closed 2022-02-08 by davidhay45