Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #89: Compiling for ATU-100 ext board

nv1o, 2022-02-15:
David, I am trying to compile an edited main.c file, version 3.2.  I downloaded MikroC PRO for PIC, ver 7.6.0.  I want to create a new hex file for the pic, but it's not working for me and I'm don't know how to get it done.  I tried creating a new project, but I'm not sure which files to include.  I've tried them all, but am getting errors like "Undeclared identifier 'ADC_Get_Sample' in expression in the main.h file, when I execute Build.  What are the steps to accomplish this?  Will I end up with the hex file to program the pic with?

73, Craig

Dfinitski, 2022-02-15:
Hi. You have to turn on the ADC library in the library manager and also couple other libraries. You will see. Open the library manager at right.

nv1o, 2022-02-15:
Spasibo!  I will give it a try.  MikroC - eto li russkaya kompaniya?

nv1o, 2022-02-15:
David, I got it to compile without errors.  How do I now generate the hex firmware file for the pic?
NV1O

Dfinitski, 2022-02-15:
That is the Poland's company.
Hex is generated. See the project folder, same name as the project

nv1o, 2022-02-15:
Yes, I looked there, but only found the 1K hex file. It generated many other files, .c.ini, .cfg, .cp, .dbg, etc.

nv1o, 2022-02-15:
That didn't work, so I redid the build with just the main.c and the 3 .h files and got this error:
1 355 Redefinition of 'main [ATU.c] '. 'main' already defined in 'main.c' ATU.c

Dfinitski, 2022-02-15:
Probably your MikroE is not activated. That is not free software .

nv1o, 2022-02-15:
No, it's not activated.  Strange it doesn't give a disclaimer on the screen.  OK, I'll see what I can do.
Спасибо
Craig

nv1o, 2022-02-15:
Got the activation code in, but same 'main already defined' error.   Спасибо за помощь.  I am ready to give up on it.

73, Craig

Dfinitski, 2022-02-16:
You have declared main function second time. ATU.c I don't know what is this. Is this your written file ?

nv1o, 2022-02-17:
ATU.c is a very small file that MikroC produces when I run build. This is what's in the file:
void main() {

}
I don't know what it's for, but if I exclude it from the build. I don't get the second function error.  I do get the following report:

0 1 mikroCPIC1618.exe -MSF -DBG -pP16F1938 -DL -O11111114 -fo8 -N"C:\ATU100\Projects\ATU100.mcppi" -EH"C:\ATU100\atu_100_fw_00.ihex" -SP"C:\Mikroelektronica\mikroC PRO for PIC\Defs\" -SP"C:\Mikroelektronica\mikroC PRO for PIC\Uses\P16_Enh\" -SP"C:\ATU100\Projects\" -SP"C:\ATU100\" -IP"C:\ATU100\" "main.c" "oled_control.c" "pic_init.c" "backup.mcl" "main.mcl" "oled_control.mcl" "pic_init.mcl" "__Lib_Math.mcl" "__Lib_MathDouble.mcl" "__Lib_System.mcl" "__Lib_Delays.mcl" "__Lib_CType.mcl" "__Lib_CString.mcl" "__Lib_CStdlib.mcl" "__Lib_CMath.mcl" "__Lib_MemManager.mcl" "__Lib_Conversions.mcl" "__Lib_Sprinti.mcl" "__Lib_Sprintl.mcl" "__Lib_Time.mcl" "__Lib_Trigonometry.mcl" "__Lib_Button.mcl" "__Lib_Keypad4x4.mcl" "__Lib_Manchester.mcl" "__Lib_OneWire.mcl" "__Lib_PS2.mcl" "__Lib_Sound.mcl" "__Lib_SoftI2C.mcl" "__Lib_SoftSPI.mcl" "__Lib_SoftUART.mcl" "__Lib_ADC.mcl" "__Lib_EEPROM.mcl" "__Lib_FLASH_rwe.mcl" "__Lib_I2C_c34.mcl" "__Lib_PWM_c216b0a4.mcl" "__Lib_SPI_c345.mcl" "__Lib_UART_c67.mcl" "__Lib_PortExpander.mcl" "__Lib_CANSPI.mcl" "__Lib_CF.mcl" "__Lib_GlcdFonts.mcl" "__Lib_Glcd.mcl" "__Lib_LcdConsts.mcl" "__Lib_Lcd.mcl" "__Lib_RS485.mcl" "__Lib_S1D13700.mcl" "__Lib_T6963C.mcl" "__Lib_SPIGlcd.mcl" "__Lib_SPILcd.mcl" "__Lib_SPILcd8.mcl" "__Lib_SPIT6963C.mcl" "__Lib_EthEnc28j60.mcl" "__Lib_EthEnc24j600.mcl" "__Lib_TouchPanel.mcl"  
0 1139 Available RAM: 1016 [bytes], Available ROM: 16384 [bytes]  
0 122 Compilation Started oled_control.h
3 393 'Soft_I2C_Scl' Identifier redefined oled_control.h
4 402 ; expected, but 'at' found oled_control.h
4 393 'Soft_I2C_Sda' Identifier redefined oled_control.h
4 371 Specifier needed oled_control.h
4 396 Invalid declarator expected'(' or identifier oled_control.h
5 402 ; expected, but 'at' found oled_control.h
5 393 'Soft_I2C_Scl_Direction' Identifier redefined oled_control.h
5 371 Specifier needed oled_control.h
5 396 Invalid declarator expected'(' or identifier oled_control.h
5 393 '' Identifier redefined oled_control.h
6 402 ; expected, but 'at' found oled_control.h
6 393 'Soft_I2C_Sda_Direction' Identifier redefined oled_control.h
6 312 Internal error '' oled_control.h
0 102 Finished (with errors): 17 Feb 2022, 11:28:07 ATU100.mcppi

I'm not sure, but the problem may be from not knowing exactly which files to include in the build from the github site.  The report above was produced with ALL the files from the 1938_EXT_board_sources_V_3.2 folder.  Which files do I include?

73, nv1o

davidhay45, 2022-02-17:
MikroC expects to find the initial start routine, main() in a file with <name of the project>.C.  So I guess that your project is called ATU?
To fix this issue simply edit ATU.C and rename the main() routine to be something else, for example main_x().  That will probably fix the "Redefinition of 'main [ATU.c] '. 'main' already defined in 'main.c' " error.
However, this system is not really the place to learn how to use MikroC.  David N7DDC has been very generous in making this project open source, we (me included) should not be asking questions like this.  Just my opinion of course.

nv1o, 2022-02-17:
Well, no matter.  It just surprises me how many people take exception to others' questions about things they don't understand in these kinds of forums.  Yes, David has been gracious to try and help me and I am grateful for that. I just can't understand why I am getting all these errors.  I did edit the ATU.c file and it did get rid of the extra function declaration error, but it's still not working.  I will quit posting and just give up on this.

Dfinitski, 2022-02-17:
Hi. My project is not named ATU and does not have any ATU.c file. 
You can see all the files right here, on Github. 
Give me a link where you see ATU.c file.

Dfinitski, 2022-02-17:
Probably you have changed some thing in settings

nv1o, 2022-02-18:
Давид, хочу сказать большое Вам спасибо за помощь.  Мне жалко, что я мало понимаю об этом.  Кстати, мне очень нравится тунер!  Прекрасно работает.  Сейчас по-антлиский.  The kit I ordered had ver. 3.2 installed, but both the code and EEPROM were protected, so I used pickit3 to reprogram with your hex file and I was able to figure out how to make my oled display work again from your documentaion file.   So now, it is all unprotected and I can change parameters when I want to.  I saved off a new hex file and all is good.  So now I want to try building the hex file from your source files to see if I can get it to work, so that I would be able to make minor programming changes in the future.  I'm just not having any success to compile it.  I did not change any settings, except to rename the default project to ATU instead of MyProject.  That is how ATU.c got created by MikroC.  What I am asking is what do I need to do in MikroC to get your source files to compile properly?  Do I just create a new project, add files and libraries and do the build, or are there other steps to take? If this is too complex to work out in the forum, let me know and I will put it away for now.  If you are willing and would prefer to communicate via email, you can get my email address from my QRZ page.

С уважением, Craig NV1O

And apologies to DavidHay for me giving it one more try.

Dfinitski, 2022-02-18:
I got you. Don't rename the project, compile it as is.
Turn on in the library manager Button, ADC, EEPROM, Conversion libraries .

davidhay45, 2022-02-18:
Just add the .c files.  The .h are dealt with by the include statements. So main.c, pic_init.c and oled_control.c are all you need. If any other .c file is seen in project manager you can remove it.  And as David N7DDC says just tick the four libraries he mentions.

closed 2022-02-20 by nv1o