Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #90: PIC clock frequency

nv1o, 2022-02-22:
I noticed that when I compiled the source files, the default clock freq. was 8 MHZ. The tuner functions ran much faster, but tuning on 60M caused display and tuner errors, I suspect, because of some mixing problems with the PIC clock and the 60M frequency.  What should the clock frequency be to get it back to normal?   Thanks.

nv1o, 2022-02-22:
Just for general info, I recompiled at decreasing clock speeds and the tuner operated faster.  I put the tuner in test mode and the up down switches toggled the caps or inductors at much higher speeds, exactly the opposite of what I expected.  After different increases of clock frequency, it looks like 12 MHZ is about right.  I will test it with each band to see if there are no problems.

Dfinitski, 2022-02-22:
Should be 16 MHz