Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #93: ATU-100 EXT Display blacks out when tuning with higher power as frequencies get higher and SWR is initially larger

kostecky, 2022-05-03:
Hi there,

I have an atu-100 that works great, except when you start trying to tune at higher powers 20-100Watts when frequencies @ upper end of 14 Mhz -> 50 Mhz, the LCD display cuts out, but the tuner still works.

Is this a known phenomenon? I assume it's because of some sort of voltage spike or drop. Can I solve this with a well-placed cap somewhere?

Thanks!

kostecky, 2022-05-03:
The key points are that it tends to happen:
- at higher powers (20-100W)
- with higher frequencies where SWR is larger and more time spent switching caps/inductors to find a match
- and ONLY after a match is found and the last relay clicks

As soon as it finds the tune, the display cuts out until I power cycle, but the tuner continues to function and it does it's job.

kostecky, 2022-05-03:
Nevermind, I fixed it. I put a 100nF cap between the VCC and GND pin on the display and it's stable as a rock. I imagined that it was voltage fluctuations more than the board can handle.

Thanks!

closed 2022-05-03 by kostecky

Dfinitski, 2022-05-03:
Thanks. That is RF interference to the display controller. Also use shortest straight path for displays wires .