Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #94: Working EE Config for Amazon ATU-100

sdr-stack, 2022-05-12:
Can't backup the Amazon prebuilt/kit Malahit ATU-100's since it's read-protected and they don't use the default settings. Here's the tuned config that I used to get it working again. I also tweaked the display timeout to get rid of the whine noise and also sped up the relays since it uses 5ms parts. You can set 03 to '15' instead of 10 to use the default relay settings and set 32 to '00' to disable the screen sleep mode if you want. Also attached the entire hex file if you just want to import and flash, rename it to .hex
[Malahit-3.2-ATU-100-dot-hex.txt](https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/files/8675867/Malahit-3.2-ATU-100-dot-hex.txt)

Address:	00	01	02	03	04	05	06	07	08	09	0A	0B	0C	0D	0E	0F

00		78	05	01	10	13	01	00	00	00	00	07	00	07	00	01	00
10		00	50	01	10	02	20	04	50	10	00	22	00	45	00	FF	FF
20		00	10	00	22	00	47	01	00	02	20	04	70	10	00	FF	FF
30		00	10	01	01	00	00	FF	FF	FF	FF	FF	FF	FF	FF	FF	FF

Dfinitski, 2022-05-21:
Thanks, it will be helpfull.

Kd8dns, 2022-07-17:
And like an idiot I tried to program the PIC before I built mine. You just saved the day!