Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #95: At high power, autotune is unstable and triggers tuning sequence when SWR is low, hence damaging relays

satfan52, 2022-05-27:
Hello, I use a 7X7 ATU-1000 tuner kit bought on Aliexpress based on the EU2AV design and that is supposedly designed to work up 1-1.2 KW.

The basis functionality works quite well and I was able to tune at high power between 1.8 and 30 Mhz with ab SWR in the order of 3 to 6. 
So the firmware works well for sure

There are  however two issues:

1) When the tune is good (SWR is low <= 1.5) auto-tune wake-up for unknown reasons  and starts a tuning sequence. This happens for power levels >= 100 W. This is a major issue when running at 1KW power levels as this unsolicited tuning sequence, when run at such high power, inevitably damage or even breaks some relays. So I have to turn off auto-tune after it is over at low power (< 100W )  to avoid that auto-tunes wakes-up again after I increase the power around 800W. The higher the power, the more easily the auto-tune triggers when not needed. The hardware settings is as follow : 100W = 1.2 V on FWD . There is no "overload" alarm at high power and the tuner tunes well with 100W, no issues. The swr bridge is a classical 20:1 tandem bridge, the directivity is better than 20dB over the range 1 - 30 Mhz. All 3 control  buttons are decoupled with 1nf capacitors. So what is going wrong ? 

2) The firmware says "ATU-100" on the screen, not "ATU-1000", it is the latest version 3.1 or 3.2. I don't think the aliexperess seller loaded the version of the firmware. The SWR indication is good but the power level indicated is completely wrong, clearly over-estimated. I don't think this the right version of firmware  for this high power ATU. Should I upload the version of the firmware with high power settings and a 20:1 ratio for the swr bridge ? Is this going to make a difference in the way auto-tune works ? 

Thanks
Regards
Peter

Dfinitski, 2022-05-27:
Hi. 1nf coupling may be not enough.  Try at least 100nf x 2.
When you increase a power , SWR could increase too and exceed the triggering level 0.3 by default.
You can make the trigger level higher or start the tuning process with higher level close to working level.
But first of all you should resolve the problem with wrong power measurement. Power and SWR should to be shown in 10% tolerance. 
Check what is tandem coefficient was set in this current firmware and correspond to real turn ratio. Check diodes on the tandem or swap them to known kind.

satfan52, 2022-05-28:
Hello David, thanks for the reply.
The seller has now sent me the correct version of the hex file. After I loaded it in the PIC16F1938,  the problem is now  completely gone. Auto-tune is now stable at 1.2 kw  (about 3.28 V on FW after tuning the Potentiometer) on 14 Mhz and there is no more unexpected wake-up. The correct power is also indicated on the display. So the problem is solved as far as I am concerned. Regards. Peter

closed 2022-05-28 by satfan52