Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #96: missin MCLR pin on my board

g14n7uca, 2022-07-14:
Purchased last week
The MCLR pin in my board is not soldered so oly 4 pins came out to connect it to pic programmer.
Suppose I have to solder one pin and then connect it to pic it 3...isn't it?
Thanks

pimol, 2022-07-14:
https://www.dropbox.com/s/jcyapzbwugt5avn/image001%20%281%29.png?dl=0

Pietro I2OIM

g14n7uca, 2022-07-14:
Thanks pietro,
I meant, miss the phisical pin on the pcboard

[image]

pimol, 2022-07-14:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/96/1184068239@
> github.com>

MCLR has to be connected with a 4.7k to VCC, as indicated. Not important
that there is no pin on the  board.
Good luck

Pietro I2OIM

pimol, 2022-07-14:
Sorry, I'm not clear.
You have to put the pin in the board.

Pietro

g14n7uca, 2022-07-14:
...and connect it to the pic board I suppose, that was the question.

pimol, 2022-10-11:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/96/1184078795@

Yes, you have to complete the connector. I had the same problem.

Pietro I2OIM