Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #97: mini ATU60-5X2 - L1 and L2 relays are constantly energized, tuning is suboptimal as 3 inductors play in instead of 5

satfan52, 2022-07-17:
Hello,

I have build two N7DCC ATUs,  a QRO ATU-1200 (7X2) that works really great up to 1.2 KW, and a QRP ATU-60 (5X2) that has a strange error and does not work very well.

On my ATU-60, the two relays corresponding to L1 and L2 on the schematic diagram are constantly energized, this corresponds to PIN 22 and 23 of the PIC16F1938. So in fact the tuner only "plays" with L3, L4 and  L5 during any tuning sequence. As L1 and L2 inductance represent 330 nh, the tuning is clearly sub-optimal especially on high bands. 

I have replaced the two 2N7000, the two relays and here is no chance, pin 22 and 23 of the PIC16F1938 jump to 5V after I power on the ATU-60 there is nothing I can do to prevent that. I also verified that there is no short-circuit. I even have replaced and reprogrammed the PIC16F1938 and there is absolutely no change.

I came to the conclusion that this is a software programming issue as I tried with the HEX file available from the site for the 5X2 model as well as with the original HEX file provided by the seller. The problem is the very same, pin 22 and 23 jump to 5V after I turn on the device and stay like that all the time, it does not matter what HEX file I use. I also tried with a 7X2 hex file and it is the same story. 

I am using a pickit2 and the pickit2+ programer to write to the 16F1938 pic, could this be the explanation for what I am seeing ? Should I have used MPLAB 6.0 and a PICKIT3 instead ? What is your view on this ? Thanks . 

Regards. Peter/on7yi

Dfinitski, 2022-07-17:
Hi, Peter
All L relays should be energized if rested, because this shortening infictances.

satfan52, 2022-07-17:
Thanks I understand I will update the title. The point is that when launching a tuning sequence L1 and L2 remain energized all the time so the tuner plays with L3, L4, L5 only.

Dfinitski, 2022-07-17:
Start your tuner in manual mode and check again. Power up with pushed down buttons bypass and Auto. If it still there, change associated transistors or change the pic controller.

satfan52, 2022-07-17:
Thanks. I will change the pic controller. I have a PCB with a PIC16F1398-I/SS (SSOP package). These chips are out of stock with Mouser and Digikey and became very expensive (more than 6USD! ) to procure on ebay/aliexpress/taobao. Is there an alternate microchip device I could use ? I checked the PIC18F2520 but it is not available in SSOP package unfortunately :(

Dfinitski, 2022-07-18:
No, I don't know what is alternate pic. 6 usd is not so expensive on my mind.

closed 2022-07-18 by satfan52