Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #98: ATU 100 EXT board (Fw ver. 3.1) LCD shuts down during tuning but system continue to work

Rentiu30, 2022-07-17:
Hello! I just got my ATU 100 mini from Aliexpress and I got it with a problem. 

Preconditions: 
- prepare setup for 10m band (power around 20W)
- ATU connected between TRX and ANT
- both transceiver and ATU are connected on the same power supply

Procedures
1) power on the unit
2) hold tune button
3) press PTT of transceiver
4) hold tune button
5) press PTT of transceiver
6) power off + power on

Expected behaviour:
1) boot up OK, all the infos are displayed on LCD
2) SWR=TUNE show on LCD
3) start to hear relays working + LCD values updated
4) SWR=TUNE show on LCD
5) start to hear relays working + LCD values updated
6) system turns off and once turned back on the LCD values indicate last tuned values

Observed behaviour:
1) boot up OK, all the infos are displayed on LCD
2) SWR=TUNE show on LCD
3) start to hear relays working + **LCD shut off**
4) LCD is off but relay reset sound is heard
5) start to hear relays working, LCD still off
6) system turns off and once turned back on the LCD values indicate last tuned values

It looks like the LCD is powering off during tuning and not recovering after. It require a system restart. While LCD is off the system can take commands from the buttons and react based on them.
But not everytime the ATU behave like this. I can reproduce the problem 80% of the times, in the rest the ATU behave as expected

The ATU is looking like the one in the picture.
[image]

I am trying to figure it out if this behaviour is a SW or HW issue.

Thanks in advance,
'73 de yo8czo

nanko55, 2022-07-18:
Hi, did you try to use a common mode choke between ATU and ANT?
73's Nanko, PA0NVY

Rentiu30, 2022-07-18:
Hello! Yes, I have, and unfortunately the behaviour is the same.

Dfinitski, 2022-07-18:
Hi. Open the box and watch inside. Most likely long wires to lCD touching some coil.

Rentiu30, 2022-07-19:
I will open up the box and longer the wires to see if the problem still occur. Thank you!

pimol, 2022-07-19:
> ID: <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/98/1188672189
> @github.com>

Put a small chock on these wires

Pietro (mobile)

Rentiu30, 2022-07-19:
I'll provide feedback once done. Currently I am not near my workbench. Many thanks!

Rentiu30, 2022-07-25:
Unfortunately nothing helped. I closed everything together and store it like it is. The unit is a back up unit.
If someone have the same issue, this topic might be a good place to continue the discussions.
I will check for updates or come back if I'll manage to fix it.

Many thanks for the proposals. '73!

GitJiggyWiddit, 2023-07-07:
Hello! 

I'm having this exact same issue. It only seems to happen on 10m with more than ~50W of power. Between 45-50W the screen just flickers but exceeding 50W makes it flicker then go out completely. Like the OP, the ATU appears to continue working. I'm wondering if this is some kind of interference issue. I have ferrite beads on the jumper to the panel mounted SO-239 socket and I also have a ferrite bead choke (many ferrite beads in a chain) where the feedline connects to the balun on my end-fed. 

Here are the relevant components in use:
SDR: https://elekitsorparts.com/product/trx-duo-sdr-transceiver-dual-16bit-adc-with-zynq7010-2tx-2rx-ddc-duc-sdr-red-pitaya-power-sdr-sdr-sdr-console/
TR Switch: https://www.ebay.com/itm/193651220379
PA: https://www.60dbm.com/product/100w-hf-50mhz-rf-power-amplifier-rd100hhf1-based-13-5v/
LPF: https://www.60dbm.com/product/100w-hf-6m-lpf/
ATU: https://www.60dbm.com/product/atu100-7x7-n7ddc-atu-antenna-tuner-oled-0-91-display/
Antenna: 40m end-fed

Any help would be greatly appreciated!

Rentiu30, 2024-04-11:
Thank you to all the colleagues who posted.

I found a bad ground point not soldered well. It's close to the board on the display side - the solder point of the cable was glued to the board. After a resolder, the shutdown of the display lowered in occurences.

However, from time to time this still happens but I'm guessing this is due to the antenna return loss during TX. The RF return loss might disturb the sensitive controllers and cause this to happen. I would say, if the antenna (before tuning) SWR is higher than 1.7, this behaviour of the ATU could be seen again. The TRX is safe but the ATU might still be influenced from the return loss of the antenna.

It's just a guess, because I've noticed on my other transceiver with internal tuner that my surrounding appliances get disturbed after 50W of RF power, on a band where the antenna naturally is close to 2:1 SWR. The tuner does the job well but the return loss influences some electronic equipments around me during TX (disconnection of CAT interface to the PC, LCD monitor goes off and on, etc.).

If this could be the case, I would isolate somehow the ATU unit by surrounding it with copper foil, add some chokers or solder additional filtering capacitors.

If this doesn't make any sense, as I'm not an expert but just commenting out from what I've noticed, please correct me or provide a different view.

Stay safe, 73!

ormopl, 2025-12-24:
> I would like to see a picture of the change made by ormopl
> 
> All the best from OV6O and Merry Christmas 25

Remove the enclousure of the ATU and send me the photo of it. I will explain it

Domino-commits, 2025-12-25:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/98/3690383927@

Hey Artur

thank you very much for helping me.

here are the pictures

best regards

Martin
OV6O

ormopl, 2025-12-25:
Sorry, u didnt upload the photos. perhaps you have to do it directly on github

ormopl, 2025-12-25:
Yes, thank you.
As i can see, you have totally different verion of the ATU. It seems to be more... modern :)

I would:

1. See if the issue exists event with the USB-C plugged in.
2. See if the 12V voltage is present during the operation, or only with the USB-C connected.
3. Measure the 5V voltage during the ATU tuning and see how the voltage drops.
4. In the second picture i can see that there is a non-populated place for the capacitor. Measure what voltage exists on these pads and eventually solder the Capacitor - say 470uF/16V and see if the problem still exists. 

Do u have any potentiometers in the bottom side of PCB ?