Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #99: Question about the tuning algorithm

davidhay45, 2022-07-26:
If the software version 3.2 in the routine sub_tune() there is a line 

if(SWR<200 & SWR<swr_mem & (swr_mem-SWR)>100) return;

this appears to say that if the SWR is less than 2 and the tuning so far has reduced the SWR by more than 1 then do not try the "other side".

Could you possibly explain what this is doing?

Thanks

Dfinitski, 2022-07-26:
Hi. It means
If SWR after tuning less than 2.0 and less than previous trying(with other side C connection) and  SWR less than previous trying for 1.0, this trying is successful.

davidhay45, 2022-07-26:
Thanks for the reply.

So if the SWR was 2.9 before this tune and 1.85 afterwards then the tune would stop?  Or is some other logic having an effect?

Dfinitski, 2022-07-27:
This logic determines wich C position is better in case if both 2 tryings where not perfectly successful.

almost-everyone, 2023-03-28:
Hi there, I'm new here and was looking at the sources but got lost looking at the tuning code in "main.h", there's a variable named "L_invert" which if I'm not wrong, controls the L/C or C/L configuration, but apparently such a variable is never toggled, even if the function "set_ind" uses it; am I missing something ?

Dfinitski, 2023-03-28:
Hi 
L_invert is the cell value flag, using for inverting of L relays control in case used 2 normally closed contacts .

almost-everyone, 2023-03-28:
> Hi L_invert is the cell value flag, using for inverting of L relays control in case used 2 normally closed contacts .

Ok, see it now, thanks !!