Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #100: ByPass fail

harmi99, 2022-08-07:
I am using ATU100 with 1:5 ratio and some modified cells, my 80m delta loop is quite simply adjusted to 21 and 28MHz FT8 frequencies (N-license) by ATU100 (SWR<1.4). However, something happended yesterday and I am no longer able to use delta loop at its resonant frequency around 3740kHz when connected to ATU, because ATU shows me infinite SWR after reset for 80m band, and the same behaviour even when ATU is switched off. If I connect antenna directly without ATU, I see antenna OK with low swr for 3740kHz. ATU itself is otherwise working fine, I can tune and make QSO at 21MHz and 28MHz below SWR1.3, however 3740kHz should work without ATU and it hardly tuned SWR ~1.7 with huge capacitance ~1500pF and some L. I verified all relays and they seems all changing their state, nothing seems burned out (anyhow I use qrp icom705 only). Could you pls advise where to look. Thanks, PH.

Dfinitski, 2022-08-07:
Hi. The best way connect it to nanovna and 50 ohm dummy. You will see if there is huge capacity or inductance or short.

harmi99, 2022-08-08:
David thanks, 

[image]
[image]
[image]

This is what I found (1-switched off ATU with dummy load and nanovna connected to ATU, 2-second picture dummyload directly at ATU, 3-FAST mode on(all relays supposed to be ON)
2nd picture shows some inductance is there, because maximum of green X=46.7Ohm at ~17.96MHz, but 3rd picture has some negative values (green line minimum is -265Ohm at  19.88MHz). So probably there is a combination of L and C failure, because above 20MHz capacitance pulls reactance to negative values. 

Do you have some better solution to find the issue than make shorts at L relays and unsolder C relays one by one? Thanks, PH.

Dfinitski, 2022-08-08:
Hi.
In manual mode using values 1,2,4,8,16,32,64 you can check if all the relays energase or deanergase. Also, using VNA you can check added reactance for each relay. It will take a time but I don't see any other reliable solution .