Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #101: More element

rembulan, 2022-10-12:
this is not an issue just stupid question,
first 5x5 then 7x7 what about 9x9 or 11x11 
why ?
why not

Dfinitski, 2022-10-12:
Hi. That tuning algorithm can't do more than 7x7. 
The new algorithm using in ATU-10 for now, can do that but...
We will need other MCU with more nodes and bigger case.
We will need much bigger PCB and housing due to much bigger high inductance coils.
The RF path will have much bigger own capacitance and inductance due to additional relays and big inductors. It won't work on 28 MHz or even 24 MHz. We will need to add even more relays and make this device more bigger, heavier and complicated.
It will be more smarter to have 2 small inexpensive devices - one 5x5 for higher bands and second 5x5 lower bands. 10x10 in total :)

rembulan, 2022-10-12:
I like it,
my question based from if we have more element the tuning is smoother but maybe I am wrong

Dfinitski, 2022-10-12:
I got you. What's wrong with a smoothness now ?

rembulan, 2022-10-12:
nothing I have 3 pcs 7x7, nothing is wrong work perfect with multiband rotary dipole, threebander yagi and multi wire dipole.
but If I can ask for wider range I think of inverted L antenna for 160m and 640m and or smoother I will more than happy.