Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #102: Noise

I4jxe, 2022-10-23:
HI to all.
I got an assembled cinese ATU, working automatic only, with PIC 16F1938  firmware 3.1.
When switched on it pull out noise to my Yaesu FT818 such as to made it unusable for me.
Is this a common and known issue of the project that can be solved ?
TIA es 73
Gabriele,  I4JXE

pimol, 2022-10-23:
Gabriele,

con cosa lo alimenti? Se è uno switching potrebbe essere la causa. Riesci a
provare con una batteria?
What are you using as power supply? If it's a switching it could be the
cause. Can you try with a battery?

Pietro I2OIM

I4jxe, 2022-10-23:
Grazie per la risposta.
Il mio ATU è alimentato dalla propria batteria interna dato che pensavio di utilizzarlo per attività mobile a bassa potenza. Con circa 2,5 W out accorda perfettamente (il venditore specifica 1 W minimo, ma  mi va bene così).. Quanto ai 100 W max indicati....mah. Comunque non ci provo....
Prima di rispedirlo al mittente vedrò di fare qualche prova strumentale,, ma ho la vaga impressione che il noise sia generato dal display..Pensavo alla soluzione di escluderne il funzionamento in ricezione, via modifica hw o fw... Che ne pensi ?
73, Gabriele

pimol, 2022-10-23:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/102/1288048822@
> github.com>

Ho un atu anch'io, comprato finito.
Ma il mio non accorda a 144, il limite dichiarato è i 6 metri a cui
peraltro non inserisce nulla perchè l'antenna risuona bene.
Lo hai modificato?
Io ho cambiato solo un parametro per farlo accordare con 5W, senza
modificare altro, altrimenti ne servivano 10. Quando uso i due metri lo
spengo ma l'antenna per i 2 metri usa il connettore frontale, per i 6-40
metri quello posteriore, quindi non saprei dirti se genera rumore.

Pietro (via Tablet)