Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #103: Milan Stejskal

ok1if, 2022-12-08:
I am the proud owner of an ATU-100
EXT, purchased on Aliexpress. Let me thank the author of SW for a perfect job. But what is my problem? My antenna (Long wire), for example, at a frequency of 3558 kHz, shows an impedance of approx. 352-j1060 ohm. That is, a high-impedance matter, where, by calculation from the Smith diagram, the antenna must be matched to the ground with a capacity of approx. 67pF and an inductance in series of approx. 18.8 uH. Of course, the original ATU-100 cannot do that. (max inductance is approx. 8.5uH). So I solved my case by inserting new inductances in the series: 0.1uH 0.2uH 0.4uH 1.0uH 2.1uH 4.3uH and I chose the largest inductance 8.6uH. Therefore, the ATU-100 should be able to tune this antenna. During the experiment, however, it tests both low and high impedance and usually moves with a maximum inductance of approx. 8uH. So he won't tune the antenna. I believe that it is caused by the SW and that is so that the capacities are not overloaded with voltage (they are only up to 500V). However, I do not want to use a tuner up to 100W, but only up to about 30W. So I know what's at stake and I don't mind. To fine-tune the antenna, I switch the tuner to TEST MODE and manually set the maximum inductance and capacity for high impedance. Everything works to my satisfaction. Unfortunately, such a transition is lengthy. My question:
1. If the max. inductance is limited by SW to 8uH, wouldn't it be possible to modify the program?
2. Alternatively, I have a suggestion: As soon as the tuner, even with 8uH protection, could not tune the antenna after several attempts, it could not switch to TEST MODE after these attempts, thereby changing the function of the TUNE AOUTO BYP buttons, of course with the last value of L and C set, so that I can continue tuning manually? Sometimes this would also be useful in automatic mode, when due to the large step L or C, it does not adjust to the optimal value.
I apologize for any inappropriate suggestions.

Milan
OK1IF (OL4W)

ok1if, 2022-12-08:
[image]
[image]

ok1if, 2022-12-08:
[image]

ok1if, 2022-12-08:
[image]

Dfinitski, 2022-12-12:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issue/103/issue_event/7990520406
> @github.com>

Hi, Milan
Join to our group for the conversation
https://groups.io/g/ATU100/topics

I have a new better tuning algorithm, I want to implement it soon in the
ATU-100

ok1if, 2022-12-12:
OK David
Thank you
Milan

> Od: "David Fainitski" ***@***.***>
> Komu: "Dfinitski/N7DDC-ATU-100-mini-and-extended-boards" ***@***.***>
> Datum: 12.12.2022 20:22
 
Hi, Milan
 Join to our group for the conversation
 https://groups.io/g/ATU100/topics
 
 I have a new better tuning algorithm, I want to implement it soon in the
 ATU-100
 
 On Thu, Dec 8, 2022, 10:58 AM Milan Stejskal ***@***.***>
 wrote:
 
 > Reopened #103
 > <https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/103>
 > .
 >
 > —
 > ID:
 > <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issue/103/issue_event/7990520406
 > @github.com>
 >

ok1if, 2023-03-17:
Hi David
Is there anything new about the new algorithm? I could help you as a beta tester.

73

Milan
OK1IF (OL4W)

> Od: "David Fainitski" ***@***.***>
> Komu: "Dfinitski/N7DDC-ATU-100-mini-and-extended-boards" ***@***.***>
> Datum: 12.12.2022 20:22
 
Hi, Milan
 Join to our group for the conversation
 https://groups.io/g/ATU100/topics
 
 I have a new better tuning algorithm, I want to implement it soon in the
 ATU-100
 
 On Thu, Dec 8, 2022, 10:58 AM Milan Stejskal ***@***.***>
 wrote:
 
 > Reopened #103
 > <https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/103>
 > .
 >
 > —
 > ID:
 > <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issue/103/issue_event/7990520406
 > @github.com>
 >

ok1if, 2023-09-15:
Hi David

What is new in new algorithm?

Milan
OK1IF