Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #107: ATU-100 Stuck in Bypass

Rimon105, 2023-05-24:
Hello all,
i have been using my ATU-100 for the past couple of months with no issues. however, 2 weeks ago it stopped tuning. the display shows its in bypass mode and none of the buttons are changing it. i used test mode to confirm the buttons work but when its booted normally, it again goes into bypass. this was a kit but i have gone back over my solder joints already and double checked my windings (ill triple check soon). Any sugestions?

Rimon105, 2023-05-25:
> Hi. I would check a voltage on the buttons.

Hello, i checked the voltage across the buttons. Bypass is .01v, auto is 3.8v, and tune is 4.8v.

Dfinitski, 2023-05-25:
Bypass is shorted. Check the connection.