Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #109: Possible PC control?

vader7071, 2023-05-27:
I just got my hands on a unit and I will be putting it on my FT-8 setup. However, my setup is a remote location (in my detached workshop) so being able to manually press the tune button is not an option unless I run out to the shop everytime I change bands.

Would it be possible to write a Python program to be able remotely control the unit through the USB port from the computer? I have successfully done this with an arduino and my antenna switch.

Dfinitski, 2023-05-27:
Hi. I saw some solutions on
https://groups.io/g/ATU100/topics
But the tuner can work in fully automatic mode so I don't see any reason to do a remote control.

vader7071, 2023-05-27:
Thank you, however the unit I have has only the single button on front (running fw 3.2) and the micro USB charging port. I don't have the bypass or auto buttons, and I can't find points RB1 or RB2 on the board to be able to add the 2 buttons.

I'm reading the manual, and I believe the unit is a manual operation (push the button to tune) and requires the "auto-switch" to have it automatically adjust. Am I reading this incorrectly?

Dfinitski, 2023-05-28:
Yes, most likely it is in manual mode.
You have two ways to turn it on 
1. Open device and find the button connection on the bottom of board and connect needed buttons.
2. But Pickit3 programmer on ebay and reprogram the MCU with needed cells value . With the programmer you can change others parameters also .

vader7071, 2023-05-28:
To make sure, this is the item I need?

[eBay pickit 3 programmer](https://www.ebay.com/itm/331901057989)

[image]

Dfinitski, 2023-05-28:
Yes it is.