Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #113: Remote buttons and display

nex2000, 2023-10-30:
perhaps this problem has already been discussed, I would like to install the tuner on the pole, obviously I will lose control of the display and buttons. Has anyone thought about remote control of the ATU100?

For the buttons it's very easy it is possible use a RS485 communications , for the display I don't know how to do it
Has anyone faced this variant?
Thank you

Dfinitski, 2023-10-31:
This tuner works well with no any buttons or display in full automatic mode.

MOUSE1025, 2023-10-31:
I have considered a remote tuner application as well.
Although I am not as interested in the display as I am the controls. The
thought of running a multi-conductor cable 100' gave me a
headache!
A Bias-T to power the tuner, and a wireless controller up the coax, not
through an antenna, can provide the necessary control(s) for the tuner. All
of this hardware can be
built for very little money.
My remote tuner is in a die-cast weatherproof enclosure with U-Bolt
mounting. The controls (momentary buttons, are on the bias-T. I did add a
latching feature for LED's to inform me of the status of the buttons.
I can provide additional information, such as pix and schematic. Very
simple project.
Regards,
Christopher

OneOrphan, 2025-02-20:
I have 75' of rotor cable on its way.  I'm not sure how to wire for everything. I have project boxes a-plenty.  It is a typical 6/2 cable...I think.  My unit is a ATU-350.  Like the 100's, extended board and such.

Kind regards,
Dick - KA5KKT

MOUSE1025, 2025-02-20:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/113/2670288960@
> github.com>

Forget the wire...use wireless control  with a bias-t for power.
K7VZD
TACTICALRADIOGEAR.COM

OneOrphan, 2025-02-20:
12v dc will not work with my remote antenna switch...goes to ground when not selected.

MOUSE1025, 2025-02-22:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/113/2672554172@

Are you using a Bias-T?
 Christopher

MOUSE1025, 2025-02-22:
I missed your earlier posting. There is a way to position the antenna
switch so that it is out of the DC path. I have a wireless contro, that
only requires a single DC +12V signal. The receiver/controller has a
antenna switch integrated into the design.
Christopher