Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #115: display not working after freeze

nathanielc, 2024-07-14:
I recently built the kit and it was working well. However the display stopped updating when I was using it, so I turned the device off and on. When powered back up the display no longer works. The tuning function does still work, I can here the relays clicking and can measure a low swr though the tuner.

I bought a replacement display thinking it was a defective unit but the new display doesn't work either. Any ideas what to check next? Would reflashing the pic be of any use?

I can confirm the display gets 5v vdd when powered on. But otherwise don't have tools to check the clock or data pins.

Thanks

Dfinitski, 2024-07-15:
Hi. If you have a simple multimeter, check the SCL and SDA lines, what is voltage there . When the device is powered off, check a short to ground, Vdd or each other.

nathanielc, 2024-07-15:
When the device is powered on voltage on each pin is:  

VCC: 5.00V
SCL: 3.75V
SDA: 0.22V

When the device is powered off checking short to ground the SDA pin has only ~175 ohms to ground. All other pins are not shorted. Is this expected? Thanks for the debugging direction.

Dfinitski, 2024-07-15:
Seems like SDA port on the PIC has burned.

nathanielc, 2024-07-15:
Ok I'll try to replace it, thanks for the help. Found the part here https://www.mouser.com/ProductDetail/Microchip-Technology/PIC16F1938-I-SS?qs=DAV8Y8VhPTJLSX3OKAfRPQ%3D%3D&mgh=1