Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #116: Test mode

ascarons22, 2024-09-29:
Does anyone have any idea why the capacitance test is not possible in test mode? The inductance test is no problem. I also deduce that this error causes the ATU to not work. Any ideas?

Dfinitski, 2024-09-30:
You can change L or C and also C connection point in the test mode.

ascarons22, 2024-09-30:
What is that point?, because I have been trying to make the tuner work for several days and it does not work, I have the feeling that the pic does not work properly.

Dfinitski, 2024-09-30:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/116/2382657229@
> github.com>

Short and long pressing of the middle button changes what will you change
with the left and right buttons. L or C or C connection point

ascarons22, 2024-09-30:
Yes, I know that point, but the capacitor test does nothing, only the inductor test works at the same time as the relays sound, the tuner is not activated either, so I deduce that it is a problem with the pic, lower power to 1w and it does not activate.

Dfinitski, 2024-09-30:
I don't know what are you meaning under "capacitor test" and "inductors test" . The re is only one manual (test) mode where you can check every capacitor and every inductor.
Also, there is the fast test when all the relays turned on.
What mode are you using and watch the problem ?

ascarons22, 2024-10-01:
The problem is that when pressing ptt on a kenwood ts-430s the atu 100 does nothing, it does not tune, version 3.2.

Dfinitski, 2024-10-01:
Ok, the power and swr values on the display are correct ?

ascarons22, 2024-10-01:
NoNo value, pulse ptt and shows nothing,The LCD shows 0, ptt pulse, shows 0, no relay sound.

Dfinitski, 2024-10-02:
Obviously, if no power so no ATU reaction on nothing. Try to figure out why the power is not going to the Stockton measuring bridge.

ascarons22, 2024-10-02:
Ok, today I will check, my equipment is a kenwood ts-430s, it is old but it works, so I deduced that it was the pic, but I will check the reflection bridge as you say, the only different thing that mounts 51 ohm resistors instead of 49.9, I will check it and inform you if I have advances.

Dfinitski, 2024-10-02:
Are you sure not installed 0.5 OHm ? What is the marking on them ?

ascarons22, 2024-10-02:
Resistor code 510

Dfinitski, 2024-10-02:
R510 ? Or 510 without R ? Better check them with a multimeter just in case. Also double check where is an input and output on the board .

ascarons22, 2024-10-02:
510 Without r, with multimeter it behaves as a short circuit, 0.3 ohm

Dfinitski, 2024-10-02:
Marking is good, but every modern multimeter is able to measure 50 ohm. You have to desolder them first,  if they on board, they will be in short circuit, it is normally.

ascarons22, 2024-10-02:
Tomorrow I will try, with other diodes it doesn't work either, but the tip of the soldering iron touching ground causes it to measure something, but not when pressing PTT. Tomorrow I will replace resistors, I have another kit on the way, until then I will keep you informed of my progress.

ascarons22, 2024-10-10:
> La marca es buena, pero cualquier multímetro moderno puede medir 50 ohmios. Primero hay que desoldar los cables, si están en la placa, se producirá un cortocircuito, es normal.

Hello, I did some tests and everything is ok, I tried other diodes, resistors, changed the pic and nothing at all, I ordered another atu 100, I mounted it and the same situation, so I deduce the problem is somewhere else. Is there any way to tune manually using an extra button? Regards

ascarons22, 2024-10-10:
solved, you can call me a noob, when making the patch cord to connect the tuner to the radio I made a short and that's why it didn't work, thanks for your help.

closed 2025-10-05 by ascarons22