Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #117: Voltage values

OK1TK, 2024-11-19:
Hi guys, juset interesting. What are the voltage values or voltages ranges on R1, R2 and then on the RA0/AN0
 and RA1/AN1 ?

OK1TK, 2024-11-27:
Based on the topic [heere](https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/117) I calculated thare should by ca 12V on the tandem-match output and ca 3.8V on the FWD/ RVS pins of the PIC for 150W. I hope that's correct.

Dfinitski, 2024-11-27:
Hi, yes 4.2 volts is the maximum value for pic ADC inputs

OK1TK, 2024-11-28:
> Hi, yes 4.2 volts is the maximum value for pic ADC inputs

Thank you David and the AC voltage alternativní the tandem-match I calculated ca 12V for 150W. Is that also correct ?

Dfinitski, 2024-11-28:
Yes, that is right. The maximum measuring power is about 150W when 10:1 tandem match ratio.