Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #118: AN= on display during Tx

Bruce-N6BMW, 2025-01-17:
I cannot find a description of what this metric is. Is it the actual forward or reflected power? Please explain. Thank you.

Dfinitski, 2025-01-17:
ANT - the power applied to antenna after all the loss calculations, fider loss and SWR loss. Read the cell description on the manual for more.

Bruce-N6BMW, 2025-01-17:
I get this error when trying to access the manual here in the files section: **Error rendering embedded code: Invalid PDF**
https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/blob/master/ATU_100_EXT_board/ATU_100_EXT_board_User_Manual/ENG_ATU-100_Extended_Board_User_Manual.pdf

Bruce-N6BMW, 2025-01-17:
Cell 34?

Dfinitski, 2025-01-17:
34 — cell for setting a feeder power loss ratio, the first number — integer part of decibell, second number — ten’s parts of decibell. Velue by default — 1.2 (12 writen in the cell). this value uses for counting the power delivered to antenna. The loss value can be found in the reference data for the used cable or you can measure the exact value yourself. If it is not necessary to take into account feeder losses, the value 00 should be written into the cell, then the calculations will correspond only to the mismatch losses.

Dfinitski, 2025-01-17:
New in the version 3.1
The fider loss value now is zero by default, the tuner board shows mismatch loss only. To install your actual fider loss, power on the board with pushed TUNE button, use BYP and AUTO buttons to set needed value. The value will be saved immediatelly in the long-term memory (0x34 cell for extended board or 0x32 cell for mini boards).

Bruce-N6BMW, 2025-01-17:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/118/2598879155@
> github.com>

Thank you!!!
BTW, I am using an ATU-500 with the AT-100 v3.2 board.

Dfinitski, 2025-01-17:
Same thing.
If you dont want to see antenna power you can switch it off in cell 33.