Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #119: Schematic - Swapped FWD/RVS Labeling ?

OK1TK, 2025-01-17:
Hi, I'm trying to fully understand the tuner schematic, and I've noticed that when I compare the schematic with the PCB, it seems that the labeling of the **FWD** and **RVS** signal connections to the bus near the branches **R1, D1...** and **R2, D2...** might be reversed—or am I misunderstanding something?

According to the schematic, the **RVS** signal should go through **D1** and the **FWD** signal through **D2**. This means that **two outputs** from the Tandem Match should go to **D2**, and **one output** should go to **D1**.

However, when I look at the board, it appears that **D1** is connected to **two outputs** from the Tandem Match, and **D2** to **only one**.

I'm basing this on the available **Gerber files**. Unfortunately, I only have assembled tuners, so I can't directly verify this on the PCB, but I assume it's the same.

Could someone please clarify this for me?

Thank you in advance!

[image]

OK1TK, 2025-01-17:
Thanks David for your reply. May be I just don’t see it correctly. Which one of the diodes on your pic is the D1  and is it correct that RVS goes through D1? Is it upper one or the bottom one ?

Dfinitski, 2025-01-17:
I dont know, actually it does not matter. the schematic could has mistake with RVS/FWD, i dont see any problem, you can change those assigment just making tandem match with different coil direction or wrongful connection of the voltage coiles. 
No any phasing dots on the schematic as you can see, so it is not defined where is what.

You can use free Sprint Layout Viewer to see the current PCB, it is fully working trust me.