Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #120: Connecting main board to display

RichFinnis, 2025-02-07:
I have just purchased an assembled kit, the n7dcc atu 100 extended board.
I am not sure how to connect them together.
There are 5 pins on the main board - mclr  vcc  gnd  dat  ck
The pins for the lcd (1202) are gnd  vcc  sda sci.
The gnd and vcc are obvious but I am not sure where the sda and sci from the display go to.
There are no instructions on the selling site but it references me here.
My electronics knowledge is basic to say the least, so that is why I purchased the built and tested kit.
I have tried studying the schematics to see if I can fathom anything out.
Any help appreciated - thanks in advance.
Rich GWOHDY

Dfinitski, 2025-02-07:
Hello
SDA - serial data
SCL - serial clock

RichFinnis, 2025-02-07:
Thanks for the fast reply -really appreciated. (so I assume ck on the board refers to clock).