Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #121: Maximum  input power for initial tuning

RichFinnis, 2025-02-09:
What is the minimum and maximum input powers for tuning please.

Dfinitski, 2025-02-09:
Depends on cells settings, see the manual.

RichFinnis, 2025-02-10:
David, thanks for the reply. However I realise I should not have posted 
on your site. Getting rather confused with who is who.
I purchased a pre assembled and tested board from a site in Ukraine, 
which trades as 60dbm but owned by an American living in the states.
It is sold with no documentation and refers folk to Github N7 DDC for 
further information. However unless I am able to read the installed 
firmware (which I am unable to)
I have no way of knowing what parameters  have been set for the min and 
max powers. At the moment I have had no response. Having read the manual 
I realise the importance of knowing
what the power parameters have been set to. Unless I hear from the 
manufacturer imminently I may return the board or start feeding the atu 
with 5 watts and increasing until I get the 
overload message then I will have some idea of max power.

input power for initial tuning (Issue #121)

RichFinnis, 2025-02-10:
I really appreciate your help - thank you very much.

 
 
Hello.
As I know 60dhm business officially registered in USA, but I don't have 
any relationship to that.
I have opened the manual and found needed value in the table together 
with other helpful data
Screenshot_20250210-062005.png (view on web)