Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #122: Any suggestions please

RichFinnis, 2025-02-15:
Hello  Folks,

I purchased an ATU100 7x7 N7DDC Assembled & tested board WITH 1602 LCD.

When it is powered up and connected to tx the lcd lights up but is blank.

I have purchased 2 new coax leads which have been checked for continuity etc plus have even changed the connecting wires which have also been checked for continuity.
These leads have been connected as follows -
Gnd to gnd
vcc to vcc
sda to data
scl to clk
mlr not connected

Set tx power to 5 watts and operated keyer but the display remains blank. 
I have tried operating mic with no change to display.

Not sure if I am missing something but any help or suggestions appreciated.

Thanks in anticipation.

Rich GW0HDY

Dfinitski, 2025-02-15:
Hello.
The LCD should to show info after power up. 
Probably the LCD board has a different I2C address or it is SPI version.
Ask the seller about this.

RichFinnis, 2025-02-16:
Hello,
Thanks for the reply. I will do this and await his response.

RichFinnis, 2025-03-06:
Hello David,

With regards to your last reply to me, I never got an answer from the seller.
At the moment I am not in a position to investigate the 12c address.
You question if the lcd board could be an SPI version. Is there any way of visually checking this or if I purchased an lcd board does it have a specific name so that I know i am not purchasing an SPI version? Or is this set up within the firmware?

Any comments appreciated.

Richard

Dfinitski, 2025-03-06:
Hi.
Purchase this LCD board and try to connect it.

[](https://www.ebay.com/itm/224176232486?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=Y1eA0PO-TMO&sssrc=4429486&ssuid=MqXhebTIQBu&var=&widget_ver=artemis&media=COPY)

RichFinnis, 2025-03-07:
Hi,
Thanks for swift reply.
Board ordered and will try that.

RichFinnis, 2025-03-29:
Hi All,

Board received but did not change things.

I have received full refund from Ebay, I do not have the skills to identify the problem - questions re firmware leave me with suspicions. However, if anyone in UK wants this I will send it to you provided you pay postage.
Its a pity as the soldering/construction is excellent.

Rich

davidhay45, 2025-03-30:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/122/2764225804@
> github.com>

I’ll try and diagnose the issue. How much is uk postage?

Rgds

David

RichFinnis, 2025-03-30:
> suggestions please (Issue #122)
> I’ll try and diagnose the issue. How much is uk postage? <br> <br> Rgds
> <br> <br> David <br> <br> On Sat, Mar 29, 2025 at 20:10 RichFinnis
> received but did not change things. <br> &gt; <br> &gt; I have received
> full refund from Ebay, I do not have the skills to <br> &gt; identify
> the problem - questions re firmware leave me with suspicions. <br> &gt;

David,
 
A local club member collected this before I had seen your email.
 
Thanks for your interest.
 
Rich

pay <br> &gt; postage. <br> &gt; Its a pity as the 
soldering/construction is excellent. <br> &gt; <br> &gt; Rich <br> &gt; 
<br> &gt; 
&lt;https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/122#issuecomment-2764225804&gt;, 
to this thread.Message <br> &gt; ID: <br> &gt; 
&lt;Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/122/2764225804@ 
<br> &gt; github.com&gt; <br> &gt; [image: RichFinnis]*RichFinnis* left 
a comment <br> &gt; 
(Dfinitski/N7DDC-ATU-100-mini-and-extended-boards#122) <br> &gt; 
&lt;https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/122#issuecomment-2764225804&gt; 
<br> &gt; <br> &gt; Hi All, <br> &gt; <br> &gt; Board received but did 
not change things. <br> &gt; <br> &gt; I have received full refund from 
Ebay, I do not have the skills to <br> &gt; identify the problem - 
questions re firmware leave me with suspicions. <br> &gt; However, if 
anyone in UK wants this I will send it to you provided you pay <br> &gt; 
postage. <br> &gt; Its a pity as the soldering/construction is 
excellent. <br> &gt; <br> &gt; Rich <br> &gt; <br> &gt; — <br> &gt; 
&lt;https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/122#issuecomment-2764225804&gt;, 
to this thread.Message <br> &gt; ID: <br> &gt; 
&lt;Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/122/2764225804@ 
<br> &gt; github.com&gt; <br> &gt; <br>
davidhay45davidhay45 left a comment 
I’ll try and diagnose the issue. How much is uk postage? <br> <br> Rgds 
<br> <br> David <br> <br> On Sat, Mar 29, 2025 at 20:10 RichFinnis 
received but did not change things. <br> &gt; <br> &gt; I have received 
full refund from Ebay, I do not have the skills to <br> &gt; identify 
the problem - questions re firmware leave me with suspicions. <br> &gt; 
pay <br> &gt; postage. <br> &gt; Its a pity as the 
soldering/construction is excellent. <br> &gt; <br> &gt; Rich <br> &gt; 
<br> &gt; 
&lt;https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/122#issuecomment-2764225804&gt;, 
to this thread.Message <br> &gt; ID: <br> &gt; 
&lt;Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/122/2764225804@ 
<br> &gt; github.com&gt; <br> &gt; [image: RichFinnis]*RichFinnis* left 
a comment <br> &gt; 
(Dfinitski/N7DDC-ATU-100-mini-and-extended-boards#122) <br> &gt; 
&lt;https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/122#issuecomment-2764225804&gt; 
<br> &gt; <br> &gt; Hi All, <br> &gt; <br> &gt; Board received but did 
not change things. <br> &gt; <br> &gt; I have received full refund from 
Ebay, I do not have the skills to <br> &gt; identify the problem - 
questions re firmware leave me with suspicions. <br> &gt; However, if 
anyone in UK wants this I will send it to you provided you pay <br> &gt; 
postage. <br> &gt; Its a pity as the soldering/construction is 
excellent. <br> &gt; <br> &gt; Rich <br> &gt; <br> &gt; — <br> &gt; 
&lt;https://github.com/Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/122#issuecomment-2764225804&gt;, 
to this thread.Message <br> &gt; ID: <br> &gt; 
&lt;Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/122/2764225804@ 
<br> &gt; github.com&gt; <br> &gt; <br>

davidhay45, 2025-03-30:
> 
> David,
>  
> A local club member collected this before I had seen your email.
>  
> Thanks for your interest.
>  
> Rich
>  
> ---

No probs. I was going to try a known good lcd and maybe load some test firmware. 
BW

KDMcMullan, 2025-06-03:
I have a suggestion: change the title of this thread to something reflecting the subject. Eg: "LDC lights up but is blank".