Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #123: Bargraph / Guage?

KDMcMullan, 2025-03-28:
I've not built this yet, but from what I can tell, when displaying the power, there is the number (in Watts) which changes on the screen. Would it be possible to replace this with a bar which changes size depending on the power, or a graphical needle?

Maybe the update rate of the OLED would be too slow. It was just a thought.

Dfinitski, 2025-03-28:
Hi, that is not good idea, users expect to use this as a measuring gauge with good accuracy.

KDMcMullan, 2025-04-03:
I propose it as an option, rather than an absolute alternative. Some might say that a graphical SWR is more use when trimming an antenna than a  number which changes on the screen. the digital presentation could be present at the same time.

Here's an even better idea: if the power / SWR were presented graphically as a chart against time, then the user could see the results of their trimming.

Dfinitski, 2025-04-03:
Hello.
This is not a SWR meter, this is an autotuner. Display as an option.
Also, this project is completely finished and I wont make any changes.
You can look on chinese ATU-100M with an alternative display graph.

https://www.aliexpress.us/item/3256808367951450.html?spm=a2g0o.productlist.main.22.946eXV2xXV2x1n&algo_pvid=84e14f4a-8ad7-4c8f-bba0-600197c6ef22&algo_exp_id=84e14f4a-8ad7-4c8f-bba0-600197c6ef22-21&pdp_ext_f=%7B%22order%22%3A%221%22%2C%22eval%22%3A%221%22%7D&pdp_npi=4%40dis%21USD%21107.56%2172.07%21%21%21782.80%21524.48%21%40212e509017437103050895863e57de%2112000045688898294%21sea%21US%21131652596%21X&curPageLogUid=WBLehyUPjn8R&utparam-url=scene%3Asearch%7Cquery_from%3A