Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #124: Low power readings

KC3TDR, 2025-04-19:
I purchased a fully assembled ATU-100 EXT V3.2 by N7DDC (no battery) sold by Malahit. This unit has Tune, Auto, and Byp buttons. The unit works very good for SWR and tuning, but reads 20% low on power consistently on all bands. I tested at 10W and 50W. Is there a power display calibration adjustment/procedure? Any help you can provide would be appreciated.
Thanks,
Frank KC3TDR

Dfinitski, 2025-04-19:
Hi,
No, but you can try to change the Stockton bridge diodes with original bat41.
It should work out of the box with 10% tolerance.
Also you can change a resistive decider ratio, see the schematic.

KC3TDR, 2025-04-19:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/124/2816836003@
> github.com>

Hi,
Thanks for your feedback. Close the issue. I'll adjust components as
necessary.
Frank KC3TDR