Dfinitski/N7DDC-ATU-100-mini-and-extended-boards #127: overload at 690w

hendrixgr, 2025-05-31:
Hi any guidelines of what I need to change in order to be able to measure the full 1kw?
Right to now with 800 watts in it shows overload and 690 watts.
I suspect that the two transformers of the tandem match swr bridge need more turns...

rhdpcollins, 2025-08-19:
The forward power toroid in my unit also fried, but it was due to the primary winding, i.e., the wire through the hole in the doughnut.  At 690 Watts CW, the wire heated to the point where it melted the insulation.  Then the wire arced to the board, cracked the toroid, and filled the room with smoke.

hendrixgr, 2025-08-19:
> <Dfinitski/N7DDC-ATU-100-mini-and-extended-boards/issues/127/3201400024@
> github.com>

Yes that was what I meant to say, the wire did fried as in yours and it
damaged the toroid permanently without visual clues in my case.
Chris.